MichaelGrupp/evo #648: The issue of maintaining consistent coordinate axis ranges

teririii, 2024-03-28:
Hi，recently, I have encountered some plot issues while using EVO,I want to know how to make the coordinate axis display range of Ape or trajectory result graph consistent.
For example
I ran  a algorithm1 on dataset ，use evo_ape and plot a figure A, 
also ran algorithm2 on dataset, plot figure B,
**I hope FigA and FigB have the same y-axis range like both are 0-0.5,or x-axis both are 0-0.5, how should I set it up**

thank you，I hope to receive your answer

MichaelGrupp, 2024-04-03:
With the [Qt5Agg plot backend](https://github.com/MichaelGrupp/evo?tab=readme-ov-file#dependencies), you can use this from the plot window:

[image]

and then here:

[image]

teririii, 2024-04-14:
Thank you for your quick reply.
But I don't have this option，this is mine
[image]

My system is Ubuntu 18.04 ，opencv3.4.1 eigen3.2.0，pangolin0.5，python 2.7.17

MichaelGrupp, 2024-04-14:
This only works with PyQt5, as mentioned.

You'd need to install it (either `apt install python3-pyqt5` or `pip install PyQt5`, depending on your setup) and then:
```
evo_config set plot_backend qt5agg
```

teririii, 2024-04-15:
Thank you for solving my question,I didn't notice it before.Thanks again

closed 2024-04-15 by teririii