MichaelGrupp/evo #652: Deformation of trajectory when performing alignment

astef3, 2024-04-04:
Hi, 

firstly I would like to start with saying that you've done an outstanding job with this evaluation tool!
I am quite new with the process of evaluation and your tool, so forgive me if I ask something silly. :) 

I have tried using your tool for evaluation on my own datasets. My data is in format of rosbag, where the on trajectory is from type /nav_msgs/Odometry and the second on is tf data. Now the thing is that I want to align the two trajectories so I can perform the comparison. Non aligned trajectories and the data associated with them is shown in the pictures below:

[image]
[image]

However, when I try to align them, the tested trajectory get deformed in comparison with the reference and I see that the number of poses for the tested trajectory is now lower, data and results shown below:

[image]
[image]

I would be really grateful, if you could give me your opinion on what is the possible cause for this and suggest any guidelines on how can I fix my issue. Now in my opinion, a thing that seamed weird to me is the path length in the non aligned trajectories, there is huge difference between the reference and the tested algorithm. In my opinion, this is maybe because in some situations the LiDAR odometry has some problems and it goes "back-and-forth" (the zoomed part in the image bellow)

[image]

Best regards,
Aleksandar

astef3, 2024-04-09:
Fixed it! Found error in my data. Now everything works nice.

closed 2024-04-09 by astef3

MichaelGrupp, 2024-04-09:
> firstly I would like to start with saying that you've done an outstanding job with this evaluation tool!
I am quite new with the process of evaluation and your tool, so forgive me if I ask something silly. :)

Thank you! There are no silly questions :smile: 

One thing to add maybe: if you "see that the number of poses for the tested trajectory is now lower", it's probably because the trajectories are matched based on their timestamps before alignment. If there are not enough matches, it can introduces gaps. You can see how many poses are matched when you run it with `--verbose` / `-v` logging. See also: https://github.com/MichaelGrupp/evo/wiki/evo_traj#temporal-alignment

astef3, 2024-04-11:
Yes that was exactly the problem. Now we fixed it and its much better! Thank you for responding.