MichaelGrupp/evo #657: evo_traj with geometry_msgs/TransformStamped

magixn, 2024-04-23:
As said in wiki 'evo only reads the trajectory', the message with type _geometry_msgs/TransformStamped_ seems to be treated in _evo_traj_ the same as _geometry_msgs/PoseStamped_, but they are quite different in real. I think this may cause some misunderstanding.

MichaelGrupp, 2024-04-23:
PoseStamped and TransformStamped have different semantics, but both can describe a trajectory in 3D space. So there's no reason why evo should not allow to load trajectory data if it's stored in one of the two formats. In the end, the interpretation is user/use-case specific. Same goes for odometry trajectories.

What would you like to clarify in that regard in the documentation?

magixn, 2024-04-23:
I think I may get it. To describe a trajectory, _TransformStamped_ stores transform from G to [X+1], not typically transform from [X] to [X+1]. Is it right?

closed 2024-04-24 by magixn