MichaelGrupp/evo #660: How to record the ros bag file to use EVO

sahilsaqi, 2024-04-30:
i am using Ubuntu 20.04 LTS

RplidarA1 , IMU (mpu6050), in raspi -4 

i have recorded my ros bag file last time and when i tried  to ``./record_tf_as_posestamped_bag.py  --lookup_frequency 15 --output_topic run_1_pose --bagfile run_1.bag map base_link`` 

it is just recording with irregular data inside  not providing right data, 
i think the problem could be with my ros bag file 

my ros bag file includes topics like ``/scan``, ``/tf``, ``imu`` , i don't know which topics are required for converting into ``geometry_msgs/PoseStamped, geometry_msgs/TransformStamped, geometry_msgs/PoseWithCovarianceStamped and nav_msgs/Odometry topics``

i  am trying to achieve the absolute error, and visualize the trajectory and RMSE values

MichaelGrupp, 2024-05-10:
> my ros bag file includes topics like /scan, /tf, imu

The only topic that would make sense here is /tf, if it contains what you want. Maybe you could try to load the trajectory of the frames that you are interested in? evo can load trajectories from the /tf topic, as explained in the Wiki: https://github.com/MichaelGrupp/evo/wiki/Formats#bag--bag2---ros1ros2-bagfile

For example: `evo_traj bag my_data.bag /tf:map.base_link /tf:odom.base_link`

sahilsaqi, 2024-05-13:
> Owner

Thank you for your reply! I changed my file to another one. 
I tried a lot! it throws an error, it looks like my ros bag file is not in the correct format I guess, in this file I have topics like `/clock, /imu, /scan, /tracked_pose ` (when I plot this topic only using `evo_traj bag evo_traj.bag /tracked_pose --plot` command i can see the trajectory path on 3d but  i need to compare with ground truth which i don't know how to get), `/trajectory_node_list  `

also can you please tell me do I need to change the format of the file to TUM? when we need to use a TUM format file.

here is my ros bag file 
 [https://drive.google.com/file/d/1WhDmBRDLBp96GJEo8OXpDzRy5QwUeNAW/view?usp=drive_link](url) 
 
 and the error is 
 ``PS C:\Users\AIoTLab-PC04\evo> evo_traj bag cartographer_new_001.bag /tf:map.base_link /tf:odom.base_link
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "C:\Users\AIoTLab-PC04\evo\evo\entry_points.py", line 95, in launch
    main_module.run(args)
  File "C:\Users\AIoTLab-PC04\evo\evo\main_traj.py", line 194, in run
    trajectories, ref_traj = load_trajectories(args)
                             ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\AIoTLab-PC04\evo\evo\main_traj.py", line 105, in load_trajectories
    trajectories[topic] = file_interface.read_bag_trajectory(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\AIoTLab-PC04\evo\evo\tools\file_interface.py", line 283, in read_bag_trajectory
    from evo.tools import tf_cache
  File "C:\Users\AIoTLab-PC04\evo\evo\tools\tf_cache.py", line 28, in <module>
    import rospy
ModuleNotFoundError: No module named 'rospy'

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)
``

MichaelGrupp, 2024-05-14:
You wrote that you're using Ubuntu with ROS noetic, but the traceback shows Windows paths...

You'll need a ROS1 installation with ROS noetic to be able to import tospy / tf2, as these modules are needed to load TF messages from bags. Normal topics should also work on Windows without a ROS installation.

sahilsaqi, 2024-05-17:
> You wrote that you're using Ubuntu with ROS noetic, but the traceback shows Windows paths...
> 
> You'll need a ROS1 installation with ROS noetic to be able to import rospy / tf2, as these modules are needed to load TF messages from bags. Normal topics should also work on Windows without a ROS installation.

This Still does not work in ubuntu environment, always showing this error 
I have used another ros file which contains the /tf !
 
``aiot@ubuntu:~/myproject$ evo_traj bag my_bagfile002.bag /tf:map.base_link /tf:odom.base_link
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/aiot/ros_slam/src/evo/evo/entry_points.py", line 95, in launch
    main_module.run(args)
  File "/home/aiot/ros_slam/src/evo/evo/main_traj.py", line 194, in run
    trajectories, ref_traj = load_trajectories(args)
  File "/home/aiot/ros_slam/src/evo/evo/main_traj.py", line 105, in load_trajectories
    trajectories[topic] = file_interface.read_bag_trajectory(
  File "/home/aiot/ros_slam/src/evo/evo/tools/file_interface.py", line 286, in read_bag_trajectory
    return tf_tree_cache.get_trajectory(reader, identifier=topic)
  File "/home/aiot/ros_slam/src/evo/evo/tools/tf_cache.py", line 164, in get_trajectory
    self.from_bag(reader, topic, static_topic)
  File "/home/aiot/ros_slam/src/evo/evo/tools/tf_cache.py", line 102, in from_bag
    msg = typestore.deserialize_ros1(rawdata, connection.msgtype)
  File "/home/aiot/.local/lib/python3.8/site-packages/rosbags/typesys/store.py", line 192, in deserialize_ros1
    msgdef = self.get_msgdef(typename)
  File "/home/aiot/.local/lib/python3.8/site-packages/rosbags/typesys/store.py", line 325, in get_msgdef
    entries = self.FIELDDEFS[typename][1]
KeyError: 'tf2_msgs/msg/TFMessage'

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)
aiot@ubuntu:~/myproject$``

sahilsaqi, 2024-05-20:
I tried to Upgraded the PKg and now its version is `v1.28.0`, and the command works fine,.

But i have a another question 
i need to compare the ground truth to estimated one. what topics are required for that 

i have recorded the ros bag file which include the topics:

`/tf`, `tracked_pose`, `,imu`, `/map`, `/scan`,

Now to generate the ground truth we need to have  serialized state of a fully optimized trajectory file (optimized.pbstream) as per this documentation [Auto generate ground truth](https://google-cartographer.readthedocs.io/en/latest/evaluation.html?highlight=optimized.pbstream#how-to) using this command 
`cartographer_autogenerate_ground_truth -pose_graph_filename optimized.pbstream -output_filename relations.pbstream -min_covered_distance 100 -outlier_threshold_meters 0.15 -outlier_threshold_radians 0.02`, 

when i searched how to make this file , i found this [Exploiting the map generated by Cartographer ROS](https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html#assets-writer)
but i am not able to use this command to generate the `.pbstream `, because it needs the horizontal and verticle links connected to base link, where as in my case i use only one lidar which gives me `/laser`, 

if you have any idea where i am going wrong please help

MichaelGrupp, 2024-05-22:
> I don't have a Ground truth topic nor have I generated it.
what i need to do for that ?

A ground truth trajectory is a reference trajectory against which you compare your other trajectories. For example, a motion capture system can be used for ground truth. Or some precise GPS for outdoors. But all of this is highly dependent on your setup, plus you anyway have to do this yourself.