MichaelGrupp/evo #667: Plot Mode doesn't work for TUM format file

sahilsaqi, 2024-05-23:
My file format is this which i am using as a trajectory and i have another file with same format which i am using for Refrence  ,
`1716384833.801 0.001 0.001 0.0 0.014 -0.01 0.001 0.029
1716384834.776 0.003 -0.001 0.0 0.019 -0.017 0.001 0.039
1716384835.801 0.003 0.003 0.0 0.013 -0.007 0.002 0.025
1716384836.8 -0.001 0.002 0.0 0.013 -0.007 0.002 0.025
1716384837.799 0.002 -0.001 0.0 0.031 0.005 -0.001 0.062
1716384838.8 -0.033 0.016 0.0 0.009 0.036 -0.045 0.014
1716384839.796 -0.081 0.22 0.0 -0.006 0.058 -0.011 -0.013
1716384840.801 0.08 0.393 0.0 -0.011 0.029 -0.038 -0.025
1716384841.773 0.319 0.379 0.0 -0.007 0.009 -0.133 -0.016
1716384842.799 0.618 0.146 0.0 0.004 0.006 -0.164 0.006`,

When i tried to plot the trajectory using this command `evo_traj tum trajectory_tum.txt --ref=trajectory_ref.txt -p`, i can clearly see the results on 3D [Image](https://raw.githubusercontent.com/sahilsaqi/evo_SLAM-/173bbe8b9fc09b2e2a57491958370303f5c93c3e/test/image.png)

But when i use the same command with `--plot_mode=xz`, Then i can see the results like this [Incorrect image](https://raw.githubusercontent.com/sahilsaqi/evo_SLAM-/main/test/Screenshot%202024-05-23%20181127.png) 

Can you please help me to plot this in xz plane

MichaelGrupp, 2024-05-23:
The data is correctly displayed. Why do you use XZ if your data is in XY?

sahilsaqi, 2024-05-24:
Thanks! I have one more question when I do `evo_traj bag run01.bag --all_topics --ros_map_yaml maps/map1.yaml -p --plot_mode=xy`, i can see the map which is not correct  but I visualize it its correct size (full size)
here is the incorrect image for reference [Image](https://github.com/sahilsaqi/evo_SLAM-/blob/main/test/Screenshot%202024-05-24%20173606.png)

sahilsaqi, 2024-05-24:
> The data is correctly displayed. Why do you use XZ if your data is in XY?

Thank you for quick reply. i was wondering to visualize it on xz plane, now i got the doubt.  
Anyways!
I have one more question when I do `evo_traj bag run01.bag --all_topics --ros_map_yaml maps/map1.yaml -p --plot_mode=xy`, i can see the map which is not correct  but I want to  visualize it its correct size (full size)
here is the incorrect image for reference [Image](https://github.com/sahilsaqi/evo_SLAM-/blob/main/test/Screenshot%202024-05-24%20173606.png)

MichaelGrupp, 2024-05-29:
Sure, you can open as many trajectories as you want using `evo_traj`. See the examples in the README and the Wiki.

closed 2024-06-07 by MichaelGrupp