MichaelGrupp/evo #670: evo_res drawing questions

HZY12345, 2024-05-30:
Hello, I have a question that bothers me. I use evo_res to evaluate the results of two trajectories. Why is the starting point and end point of one of the curves always connected by a straight line?
[image]

MichaelGrupp, 2024-06-03:
Can you please upload the file that creates this plot?

zmxiaoj, 2024-06-07:
Having same question, when using evo_res to plot different ape.
```shell
# I use these command
evo_ape tum groundtruth traj2.txt -vas -p --save_results results_ape_sim3/traj2.zip
evo_res results_ape_sim3/traj*.zip --save_table table0.csv -p
``` 
And this is result
[image]
These are relative files

[table.csv](https://github.com/user-attachments/files/15728387/table.csv)
[traj0.zip](https://github.com/user-attachments/files/15728392/traj0.zip)
[traj1.zip](https://github.com/user-attachments/files/15728397/traj1.zip)
[traj2.zip](https://github.com/user-attachments/files/15728401/traj2.zip)

MichaelGrupp, 2024-06-13:
Hello, thanks for the report. This was indeed a bug. I fixed it in https://github.com/MichaelGrupp/evo/pull/675 and it's now released together with version `v1.29.0` of evo.

closed 2024-06-26 by MichaelGrupp