MichaelGrupp/evo #677: About Drawing

110sha, 2024-06-13:
[image]
May I ask how to adjust the coordinate axis range?

MichaelGrupp, 2024-06-13:
Does this similar question here help? https://github.com/MichaelGrupp/evo/issues/648

110sha, 2024-06-14:
Thank you very much. I have also found and implemented this solution. Now I have another question that I would like to inquire about. How can I set the image to be square? I tried the command "evo_config set plod_figsize 5 5", but it didn't work as shown in Figure 1. I need Figure 2.
[image]
[image]

MichaelGrupp, 2024-06-26:
The second image you posted does not have equal coordinate axes in the plot. If you want something similar, you can disable realistic scaling with:

```
evo_config set plot_xyz_realistic false
```

...then you can stretch your plot however you like.

closed 2024-07-17 by MichaelGrupp