MichaelGrupp/evo #681: Support for setting the size of the image to save

WZG3661, 2024-06-25:
The default saved image is too small. I hope to be able to set the size of the saved image through the command line:)

MichaelGrupp, 2024-06-26:
You can change the default from the command line with:

```
evo_config set plot_figsize 15 10
```

where the two numbers correspond to the figsize setting of matplotlib in inches: https://matplotlib.org/stable/gallery/subplots_axes_and_figures/figure_size_units.html

See also https://github.com/MichaelGrupp/evo/wiki/Plotting#3-set-the-default-figure-size

WZG3661, 2024-06-26:
Thanks a lot!
if the scale of axis are equal will be better:
```Python
plt.axis("equal")
```

MichaelGrupp, 2024-06-26:
This should be the case already for trajectory plots, where did you miss equal axis scale?

closed 2024-07-17 by MichaelGrupp