MichaelGrupp/evo #686: The kitti dataformat and tum dataformat of same result,but have a different APE

goergehu, 2024-07-17:
Thank you for your tools. I run some code to have KITTI dataformat result and TUM dataformat result at same time. And I use your tools to evaluate these two results, but they have different results.

like this for kitti:
Loaded 4524 poses from: 00.txt
Loaded 4524 poses from: global_kitti.txt
--------------------------------------------------------------------------------
Aligning using Umeyama's method...
Rotation of alignment:
[[ 0.99685285  0.06100972 -0.05061831]
 [-0.06320537  0.99707494 -0.0429725 ]
 [ 0.0478485   0.0460366   0.99779314]]
Translation of alignment:
[-1.03516941  0.14592958 13.2319137 ]
Scale correction: 1.0
--------------------------------------------------------------------------------
Compared 4524 absolute pose pairs.
Calculating APE for translation part pose relation...
--------------------------------------------------------------------------------
APE w.r.t. translation part (m)
(with SE(3) Umeyama alignment)

       max	3.729264
      mean	2.107056
    median	2.112665
       min	0.372393
      rmse	2.220705
       sse	22310.241984
       std	0.701316

and like this for TUM:
Loaded 4541 stamps and poses from: ../00.txt
Loaded 4524 stamps and poses from: global.txt
Synchronizing trajectories...
Found 4524 of max. 4524 possible matching timestamps between...
	../00.txt
and:	global.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Aligning using Umeyama's method...
Rotation of alignment:
[[ 0.9979565  -0.04333859 -0.04695297]
 [ 0.04618998  0.99704083  0.06144969]
 [ 0.04415089 -0.06349287  0.99700519]]
Translation of alignment:
[13.32838423  1.40974435 -0.6254514 ]
Scale correction: 1.0
--------------------------------------------------------------------------------
Compared 4524 absolute pose pairs.
Calculating APE for translation part pose relation...
--------------------------------------------------------------------------------
APE w.r.t. translation part (m)
(with SE(3) Umeyama alignment)

       max	3.049987
      mean	1.181944
    median 1.042915
       min	0.201317
      rmse	1.342138
       sse	8149.241122
       std	0.635881

I have transform their coordinate before evaluate, the results confused me a lot, Could you help me?

goergehu, 2024-07-17:
Thank you for your reply.
I run my  code used KITTI-00, and I got results of poses. I save poses as TUM and KITTI format, but the number of poses is less than KITTI groudTruth because of Initialization. Then I delete some begining groudTruth poses so as to have the same number pose. Finally, I evaluate the APE of these two poses, they have different results, as described before.

goergehu, 2024-07-17:
by the way, when the TUM poses is evaluated in LiDAR Frame, and the KITTI poses is evaluated in CAMERA Frame(why I say transform corrodinate).

MichaelGrupp, 2024-07-17:
As said, you would need to check if your data is the same in both formats. If that's not the case, differences are expected.

> TUM poses is evaluated in LiDAR Frame, but the KITTI poses is evaluated in CAMERA Frame

This most likely means that they are _not_ the same, as the sensors are probably not in the same location on the vehicle.

closed 2024-07-30 by MichaelGrupp