MichaelGrupp/evo #687: Map Tile Baselayer in Plot

frapit, 2024-07-18:
Thanks a lot for this great tool!

**Is your feature request related to a problem? Please describe.**
It would be nice if we could interface a small map tile library such as [contextily](https://github.com/geopandas/contextily) to display a map background. I often want to plot a trajectory from a rosbag, for instance using `/tf:utm.base_link`. 

**Describe the solution you'd like**
It would be nice if I could easily add a map baselayer for the UTM (or whatever) extent if the coordinates are geo-referenced anyway.

**Describe alternatives you've considered**
This can be achieved through `--ros_map_yaml` with a rgb image which could for instance be generated with a snippet as:
```
import contextily as cx
import matplotlib.pyplot as plt

west, south, east, north = (8.0, 49.0, 8.1, 49.1)
img, ext = cx.bounds2img(west, south, east, north, ll=True, source=cx.providers.CartoDB.PositronNoLabels)

plt.imshow(img, extent=ext)
plt.axis('off')
plt.savefig('tile.png', bbox_inches='tight', pad_inches=0)
```
This at least requires the extent of the plot generated by `evo_plot`. It would be a help if this could be written to the stat file or somewhere else.

frapit, 2024-07-23:
I quickly added it to the code as it is just a oneliner for my purpose. If someone needs to do the same thing, just add the import `import contextily as cx` and then call
```
cx.add_basemap(ax_traj, crs="epsg:32632", source=cx.providers.CartoDB.PositronNoLabels, zoom=20)
```
with the respective projection (UTM in this case) and zoom level as crs string. I did the call where the ros_map is applied as well.

If you are willing to add this feature, one would need to have additional configuration options: `base_map_crs`, `base_map_source` and `base_map_zoom`. But it also needs a bunch of libraries so I also understand if you don't want too many external dependencies. Just wanted to leave it here as reference.

MichaelGrupp, 2024-07-30:
Sounds like a neat add-on, the question would be for me how you would envision to use this in the command line interface.

> If you are willing to add this feature, one would need to have additional configuration options: `base_map_crs`, `base_map_source` and `base_map_zoom`. 

I would say that it's probably easier to have a single flag, named for example `--geo_tile` and then take the three arguments from it? Maybe also having a default (configurable) map provider, so you can also just pass the crs and the zoom?

So either...

1. `--geo_tile epsg:32632 20` (+ `geo_tile_provider` in global settings)
2. `--geo_tile epsg:32632 20 CartoDB.PositronNoLabels`

I guess first one is more user friendly, as you probably don't want to specify/change the provider all the time.

> But it also needs a bunch of libraries so I also understand if you don't want too many external dependencies.

We can have optional dependencies, so this is not a blocker.

frapit, 2024-07-31:
> I would say that it's probably easier to have a single flag, named for example `--geo_tile` and then take the three arguments from it? Maybe also having a default (configurable) map provider, so you can also just pass the crs and the zoom?

Sounds good! contextily figures out a good zoom level according to the requested size if none is given, so this could be optional as well. 

> I guess first one is more user friendly, as you probably don't want to specify/change the provider all the time.

Agreed!

> We can have optional dependencies, so this is not a blocker.

Great!

MichaelGrupp, 2024-08-01:
Since I don't have georeferenced data: do you have maybe a minimal example that you could share for testing? One random file with corresponding EPSG should be enough.

The feature seems rather easy to add, but would be good to check if everything works as expected :wink:

frapit, 2024-08-05:
Sure, the file [example_traj_tf_utm_baselink.txt](https://github.com/user-attachments/files/16494355/example_traj_tf_utm_baselink.txt) produces the following output:

[image]

when calling `evo_traj tum example_traj_tf_utm_baselink.txt --plot_mode xy -p`

with `epsg:32632` as crs. (By simply adding the code line above)

I have other examples if required.

MichaelGrupp, 2024-09-06:
Thanks for the data! I added the feature now: https://github.com/MichaelGrupp/evo/commit/29c20d0f1a111b9b55896c36101a694c506c64fc (Release v1.30.0)
Tested with OpenStreetMap, CartoDB, HEREv3 (requires token).

Basic usage to get your example would be:

```
evo_config set map_tile_provider CartoDB.PositronNoLabels
```
(for providers with API token also set `map_tile_api_token`)

```
evo_traj tum example_traj_tf_utm_baselink.txt --plot --map_tile epsg:32632
```

I would like to use your data as example data / plot for the documentation, would that be fine?

frapit, 2024-09-06:
Looks good, thanks a lot!

> I would like to use your data as example data / plot for the documentation, would that be fine?

Sure, feel free to use the data and close the issue.

MichaelGrupp, 2024-09-09:
Perfect, thanks! https://github.com/MichaelGrupp/evo/wiki/Plotting#geographic-map-tiles