MichaelGrupp/evo #691: Question about RPE and coordinate system

Davidyao99, 2024-09-04:
I am having some difficulty with trying to understand the relation between RPE and coordinate systems. I have a GT trajectory and a pred trajectory, and I evaluate them using the command below:

```
RPEt = main_rpe.rpe(traj_ref, traj_est, est_name='traj', align=True, correct_scale=True,
    pose_relation=PoseRelation.translation_part, delta=1, delta_unit=Unit.frames, all_pairs=True)
RPEr = main_rpe.rpe(traj_ref, traj_est, est_name='traj', align=True, correct_scale=True,
    pose_relation=PoseRelation.rotation_angle_deg, delta=1, delta_unit=Unit.frames, all_pairs=True)
```

1) Based on my understanding of the RPE formula, the relative translation should not be affected if I change the coordinate system of the GT trajectory. However, I notice that if I apply a coordinate transform to both the rotation and translation of the GT trajectory, my RPEt and RPEr values changes. Is my understanding wrong? What could be the issue here?

2) If coordinate systems do affect the evaluation results, how should we align the GT and pred trajectory so that they use the same coordinate system? Based on my understanding of the alignment provided by evo, it only aligns based on minimizing the translation (shape of trajectory).

MichaelGrupp, 2024-09-16:
RPE is not affected as long as the translations and rotations between the poses inside each trajectory are not changed. This is the case when a rigid transformation is used.

> I notice that if I apply a coordinate transform to both the rotation and translation of the GT trajectory, my RPEt and RPEr values changes. Is my understanding wrong? What could be the issue here?

I don't know what you are doing in your code. But if you have a trajectory object and pass a SE(3) rigid transformation matrix to the [transform](https://github.com/MichaelGrupp/evo/blob/master/evo/core/trajectory.py#L162) method, it will not affect RPE.

Davidyao99, 2024-09-16:
Indeed, I realized I had a transformation bug. It is resolved and everything makes sense now. Thank you!