MichaelGrupp/evo #692: Align causes some paths to be be truncated

EmmanuelMess, 2024-09-19:
**Description:**
Aligning the origin causes paths to be truncated.

**Command:**
```
evo_traj tum -p --align_origin --plot 'sim/2024-09-18-19-33-15/gt.txt' 'sim/2024-09-18-19-33-15/stereo.txt' 'sim/2024-09-18-19-33-15/stereo+imu.txt' 'sim/2024-09-18-19-33-15/stereo+imu+gnss.txt' 'sim/2024-09-18-19-33-15/stereo+imu+wo.txt'   --ref 'sim/2024-09-18-19-33-15/gt.txt'  --save_plot test
```

**Console output:**
```

--------------------------------------------------------------------------------
name:	stereo
infos:	29 poses, 2.601m path length, 5.603s duration
--------------------------------------------------------------------------------
name:	stereo+imu
infos:	30 poses, 2.884m path length, 5.800s duration
--------------------------------------------------------------------------------
name:	stereo+imu+gnss
infos:	28 poses, 2.720m path length, 5.400s duration
--------------------------------------------------------------------------------
name:	stereo+imu+wo
infos:	28 poses, 2.750m path length, 5.400s duration
--------------------------------------------------------------------------------
name:	gt
infos:	130 poses, 11.021m path length, 12.919s duration
--------------------------------------------------------------------------------
Plot saved to test_trajectories
Plot saved to test_xyz
Plot saved to test_rpy
Plot saved to test_speeds

```
**Command:**
```
evo_traj tum -p --plot 'sim/2024-09-18-19-33-15/gt.txt' 'sim/2024-09-18-19-33-15/stereo.txt' 'sim/2024-09-18-19-33-15/stereo+imu.txt' 'sim/2024-09-18-19-33-15/stereo+imu+gnss.txt' 'sim/2024-09-18-19-33-15/stereo+imu+wo.txt'   --ref 'sim/2024-09-18-19-33-15/gt.txt'  --save_plot test 
```

**Console output:**
```
--------------------------------------------------------------------------------
name:	stereo
infos:	185 poses, 5.985m path length, 12.278s duration
--------------------------------------------------------------------------------
name:	stereo+imu
infos:	188 poses, 6.561m path length, 12.473s duration
--------------------------------------------------------------------------------
name:	stereo+imu+gnss
infos:	181 poses, 6.345m path length, 12.000s duration
--------------------------------------------------------------------------------
name:	stereo+imu+wo
infos:	183 poses, 6.403m path length, 12.133s duration
--------------------------------------------------------------------------------
name:	gt
infos:	130 poses, 11.021m path length, 12.919s duration
--------------------------------------------------------------------------------
Plot saved to test_trajectories
Plot saved to test_xyz
Plot saved to test_rpy
Plot saved to test_speeds

```

**Additional files:**
Please attach all the files needed to reproduce the error.

[2024-09-18-19-33-15.zip](https://github.com/user-attachments/files/17051754/2024-09-18-19-33-15.zip)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.30.0
* Python version shown by `evo pkg --pyversion`: 3.10.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 22.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "map_tile_api_token": "",
    "map_tile_provider": "OpenStreetMap.Mapnik",
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_enabled": true,
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

EmmanuelMess, 2024-09-19:
> The total number of poses can change with alignment because a time-based synchronization with the reference trajectory is done for each trajectory. Only the poses that match with the reference are kept. See: [Wiki: evo_traj (temporal alignment)](https://github.com/MichaelGrupp/evo/wiki/evo_traj#temporal-alignment)

Can't I just align all path starts to (0,0,0)? Is the `--align_origin` also doing  `--align`?

MichaelGrupp, 2024-09-20:
`--align_origin` does not put all paths to 0,0,0. It makes the trajectories start aligned to the first matching pose in the reference.

MichaelGrupp, 2024-09-23:
Possible yes, just transform using the inverse of the first pose. But there is no feature to do this from the command line. With the evo lib, you would need a script doing something like:
```
traj = file_interface.read_tum_trajectory_file("traj.txt")
traj.transform(lie_algebra.se3_inverse(traj.poses_se3[0]))
file_interface.write_tum_trajectory_file("out.txt", traj)
```

closed 2024-09-23 by EmmanuelMess