MichaelGrupp/evo #693: Alignment with Starting at the same Initial Pose

kameel2311, 2024-09-26:
The alignment does not take into consideration the constraint of starting the trajectories at the same pose. This is an issue as the --align would try to figure out the best un-constrained transformation that might skew the path off a more realistic alignment due to having sub-optimal path estimation from the ground truth.

Using the aligned_origin flag alongside align does not make a difference and having that as a feature would give more realistic errors.

MichaelGrupp, 2024-09-27:
I don't understand what feature you are missing. It seems like you are talking about a constraint "starting the trajectories at the same pose" that you want to fulfill to have a "realistic alignment". But I am missing an exact specification of what you want to achieve and/or an example illustrating your current issue.

> --align would try to figure out the best un-constrained transformation that might skew the path off a more realistic alignment

`--align` computes the least-squares alignment of both paths. Whether this is useful or not is completely use-case dependent, so please specify what "realistic" result is in your case.

> Using the aligned_origin flag alongside align does not make a difference

Not sure what you are trying to achieve with both flags, but `--align_origin --align` does not really make sense. Either use
* `--align` (full alignment, optionally with `--n_to_align <number>` to only align a limited number of poses)
* or `--align_origin` (aligning at the first matching pose).

Out of these options, `--align_origin` sounds like what you are looking for?

kameel2311, 2024-10-02:
Checking the main_ape.py, why is the align_origin flag in an elif and not an if statement in line 64

MichaelGrupp, 2024-10-14:
Thanks for the more detailed explanation. I see different things here:

* `--align_origin --correct_scale`: Mixing those was so far not yet considered in the evo_ape/rpe apps, but can be done. As you pointed out, this can be done by changing from `elif` to `if`.
* `--align --align_origin`: This is not supported, and also algorithmically doesn't make sense as it would break the properties of each other:
  * forcing start pose (position + orientation) to be the same
  * vs. minimizing distance over all poses

This is now handled clearer in v1.30.1, using both flags together is not possible anymore (details [here](https://github.com/MichaelGrupp/evo/commit/8f0d639b5a405bd710c9d812dab16c14299ff28d))

> idea is to have alignment and aligning the origin at the same time (constrained optimization in a sense)

This would require a completely new alignment algorithm, it's not possible to achieve with the available options.

MichaelGrupp, 2024-10-24:
> My understanding is that the pose error should start from zero and then drift.

I assume you are talking about an absolute pose error here. As you observed, these errors can quickly accumulate when aligning only at/around the start only. Errors at the start are weighted high, and the trajectory length influences the result too. That bias is one of the reasons why relative pose errors exist, because they measure drift invariant to global alignment and independent of total trajectory length.

I can't comment whether you "should" measure like that, because it depends on your use case. But I can say that absolute errors are usually not well-suited for drift comparison. 

My recommendation that should cover most of the real-world evaluation scenarios is:

* APE without alignment: evaluation of absolute accuracy (example: localization quality)
* APE + full alignment: evaluation of overall trajectory shape consistency (example: mapping quality)
* RPE: evaluation of drift (example: odometry quality, e.g. [rotational error per meter](https://github.com/MichaelGrupp/evo/wiki/Metrics#evo_rpe))

These are fair metrics that allow also to compare different results.

APE + origin alignment is also valid, but has this strong sensitivity on the quality of the first pose(s) that you need to keep in mind, so less suited to assess overall trajectory quality. APE with some new constrained alignment that minimizes around an arbitrary number of first poses while keeping the first position fixed (but not its orientation) also doesn't get rid of that problem. So I would recommend to check if a relative metric isn't better suited for the task.

closed 2024-10-29 by MichaelGrupp