MichaelGrupp/evo #699: Axis label get cut off when zooming

fdila, 2024-10-30:
**Description:**

When zooming the plot, the axis label/name gets cut off. It's just a little bug but is quite annoing.

Before zooming (everything is ok)
[image]

After zooming (x label gets cut)

[image]

The label gets cut also when saving the image directly from the UI.

[image]

closed 2024-10-30 by MichaelGrupp

MichaelGrupp, 2024-10-30:
I also saw this a few times. Since the canvas is managed by matplotlib, I quickly checked what could help there and found an easy fix (see linked pull request). With this I can zoom and resize the window in all kinds of ways and it does not cut off anything.

Patch is released in version 1.30.2 now.

fdila, 2024-10-30:
Wow, that was quick!
It is working for me as well, thank you!

commit d1cf1345 (2024-10-30): Use constrained_layout instead of manual tight_layout

constrained_layout is stable since matplotlib 3.6.0 and is better than
tight_layout.

Fixes issues with resizing or zooming plots cutting off parts of the
axes.

Closes #699
PR #700 (2024-10-30): Use constrained_layout instead of manual tight_layout
constrained_layout is stable since matplotlib 3.6.0 and is better than tight_layout.

Fixes issues with resizing or zooming plots cutting off parts of the axes.

Closes #699
commit a43e66c1 (2024-10-30): Use constrained_layout instead of manual tight_layout (#700)

constrained_layout is stable since matplotlib 3.6.0 and is better than
tight_layout.

Fixes issues with resizing or zooming plots cutting off parts of the
axes.

Closes #699