MichaelGrupp/evo #708: --plot_relative_time is not used for speed graph

pbeeson, 2024-12-05:
**Description:**
 --plot_relative_time plots all things in relative time except speed graph in evo_traj

Please give also the following information:

evo version number shown by evo pkg --version:
v1.30.31

Python version shown by evo pkg --pyversion:
3.10.12

operating system and version (e.g. Ubuntu 16.04 or Windows 10):
Ubuntu 22

did you change the source code? (yes / no):
no

output of evo_config show --brief --no_color:

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "map_tile_api_token": "",
    "map_tile_provider": "OpenStreetMap.Mapnik",
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        10,
        10
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_legend_loc": "best",
    "plot_linewidth": 3,
    "plot_mode_default": "xy",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_enabled": true,
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "whitegrid",
    "plot_show_axis": true,
    "plot_show_legend": false,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "max",
        "rmse"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

MichaelGrupp, 2024-12-06:
Fixed by https://github.com/MichaelGrupp/evo/pull/709

MichaelGrupp, 2024-12-06:
Thanks for reporting this, the patch is released in version 1.30.4.