MichaelGrupp/evo #715: evo_rpe params '-a',   '--delta' and  '--all_pairs' meaning

narutojxl, 2025-01-07:
Hello author,
Thanks for your tools firstly!
*  According to https://github.com/MichaelGrupp/evo/blob/master/notebooks/metrics.py_API_Documentation.ipynb and https://github.com/MichaelGrupp/evo/wiki/Metrics,  when using evo_ape to evaluate ref and est traj,  we must be sure  to not forget  to use with param '-a',  which will firstly align ref and est traj into the same coordinate frame, in other words, using evo_ape tool must with param '-a'.
* Q1: When using evo_rpe to evaluate ref and est traj, could we not to use  param '-a' to evaluate drift? In other words, whether to use param '-a' with evo_rpe or not, will effect the error result?
* Q2: evo_rpe metrics,  in the formula,  E_i, j = (P_ref_i.inv  * P_ref_j).inv  * (P_est_i.inv  * P_est_j),  you say  i, j are  two consecutive poses.  Param '--all_pairs' you say "overlapping relative pose pairs are used.",  What does "overlapping relative pose" mean?  If we have 5 points (id = 1, 2, 3, 4, 5) of ref and est traj, consecutive poses means i, j = [1, 2],  [2, 3], [3, 4], [4, 5]. In this case,  how to understand overlapping relative pose ? 
* Q3:  param '--delta 5 --delta_unit m' means the translational or rotational drift per  5 meter. Did you means when using evo_rpe with '--delta 5 --delta_unit m', we will segment ref and est traj every 5m  into many small segment trajs according to traj length, and then using evo_rpe metrics formula to calculate rmse for every segment  traj and then averge these rmse to get a final result?

Thanks for your help and time a lot!
Best regards
narutojxl

MichaelGrupp, 2025-01-07:
> evo_ape tool must with param '-a'

It's not a must, but recommended if you want to measure the similarity of two trajectories (e.g. with a mapping trajectory). Without alignment, you measure absolute accuracy (e.g. for a localization trajectory).

> whether to use param '-a' with evo_rpe or not, will effect the error result?

Aligning with `-a` does not affect the RPE, because it is a rigid transformation of the whole trajectory and does not affect the relative transforms of poses within the trajectory. Aligning with scale correction (`-s`) however can affect RPE, becase it can affect the distance between pose pairs when scaling.

>  What does "overlapping relative pose" mean?

Let's assume you have poses with IDs `[0, 1, 2, 3, 4, 5]` and they are all 0.5m apart from each other.

* With `--delta 1 --delta_unit m`, the RPE is calculated using these consecutive pairs:
  `[(0, 2), (2, 4)]`.
* With `--delta 1 --delta_unit m --all_pairs`, the RPE is calculated using these overlapping pairs:
  `[(0, 2), (1, 3), (2, 4), (3, 5)]`

> Q3

Yes, the overall RPE is calculated by comparing all the segments and summarizing the error in a statistic like mean, rmse etc.

narutojxl, 2025-01-08:
Thanks for your detailed, kind and timely reply.

> evo_ape tool must with param '-a'

When the ref traj and est traj is respectively in two different frames, for example, ref traj is rtk traj and est is slam traj. When we use evo_ape to evaluate error, we must use with param '-a' or '--n_to_align', right? 
 When we evaluate localization traj, the ref traj and est traj are in the same coordinate frame, so there we do not use with '-a' param.

MichaelGrupp, 2025-01-08:
Exactly, these are examples where alignment makes sense / or can be avoided. I just wanted to clarify that it's not required in all cases, it depends on your specific use case.

narutojxl, 2025-01-09:
sorry

closed 2025-01-09 by narutojxl