MichaelGrupp/evo #734: Inconsistent result from evo 1.31.1 (latest) and 1.28.0

MarkChenYutian, 2025-04-06:
**Description:**

The absolute translation error (ATE) metric of the evo package is inconsistent between version 1.31.1 and 1.28.0, with a significant difference of more than 80% (the result from 1.31.1 is 180% of the result from 1.28.0).

**Command:**

This is observed within a (relatively) large codebase, but the core part of it is how we create evo pose-path and call the evo API

```python
from evo.core.trajectory import PoseTrajectory3D

# How to create evo trajectory from our own format
  @property
  def as_evo(self) -> PoseTrajectory3D:
      positions = self.poses.translation().numpy()
      rotation_wxyz = self.poses.rotation().roll(shifts=1, dims=-1).numpy()
      evo_traj = PoseTrajectory3D(
          positions_xyz=positions,
          orientations_quat_wxyz=rotation_wxyz,
          timestamps=self.time.numpy()
      )

# Main ATE evaluation entry point
import evo.main_ape as main_ape

def evaluateATE(gt_traj: PosePath3D, est_traj: PosePath3D, correct_scale=False):
    result = main_ape.ape(gt_traj, est_traj, 
                          pose_relation=PoseRelation.translation_part,
                          align_origin=True,
                          align=True,
                          correct_scale=correct_scale)
    return result
```

**Console output:**
```
(MACSLAM) (base) yutianch@manifold:~/Documents/MAC-SLAM$ pip show evo
Name: evo
Version: 1.31.1
Summary: Python package for the evaluation of odometry and SLAM
Home-page: https://github.com/MichaelGrupp/evo
Author: 
Author-email: Michael Grupp <michael.grupp@tum.de>
License: GNU General Public License v3 (GPLv3)
Location: /home/yutianch/miniconda3/envs/MACSLAM/lib/python3.11/site-packages
Requires: argcomplete, colorama, matplotlib, natsort, numexpr, numpy, pandas, pillow, pygments, pyyaml, rosbags, scipy, seaborn
Required-by: 
(MACSLAM) (base) yutianch@manifold:~/Documents/MAC-SLAM$ python -m Evaluation.EvalSeq --spaces Results/MACVO@v2_H000_old/04_05_182858/
                ╷        ╷        ╷          ╷        ╷        ╷          ╷        ╷        ╷          ╷        ╷        ╷           
  Trajectory    │ μ_ATE  │ σ_ATE  │ RMSE_ATE │ μ_RTE  │ σ_RTE  │ RMSE_RTE │ μ_ROE  │ σ_ROE  │ RMSE_ROE │ μ_RPE  │ σ_RPE  │ RMSE_RPE  
 ═══════════════╪════════╪════════╪══════════╪════════╪════════╪══════════╪════════╪════════╪══════════╪════════╪════════╪══════════ 
  Average       │ 0.4579 │ 0.2662 │ 0.5297   │ 0.0106 │ 0.0351 │ 0.0367   │ 0.1497 │ 0.3183 │ 0.3518   │ 0.0114 │ 0.0359 │ 0.0377    
  MACVO@v2_H000 │ 0.4579 │ 0.2662 │ 0.5297   │ 0.0106 │ 0.0351 │ 0.0367   │ 0.1497 │ 0.3183 │ 0.3518   │ 0.0114 │ 0.0359 │ 0.0377    

(MACSLAM) (base) yutianch@manifold:~/Documents/MAC-SLAM$ pip uninstall evo && pip install evo==1.28.0
... (Omitted, the re-install process went on successfully)

(MACSLAM) (base) yutianch@manifold:~/Documents/MAC-SLAM$ python -m Evaluation.EvalSeq --spaces Results/MACVO@v2_H000_old/04_05_182858/
Updated outdated /home/yutianch/.evo/settings.json
                ╷        ╷       ╷          ╷        ╷        ╷          ╷        ╷        ╷          ╷        ╷        ╷           
  Trajectory    │ μ_ATE  │ σ_ATE │ RMSE_ATE │ μ_RTE  │ σ_RTE  │ RMSE_RTE │ μ_ROE  │ σ_ROE  │ RMSE_ROE │ μ_RPE  │ σ_RPE  │ RMSE_RPE  
 ═══════════════╪════════╪═══════╪══════════╪════════╪════════╪══════════╪════════╪════════╪══════════╪════════╪════════╪══════════ 
  Average       │ 0.2499 │ 0.125 │ 0.2794   │ 0.0106 │ 0.0351 │ 0.0367   │ 0.1497 │ 0.3183 │ 0.3518   │ 0.0114 │ 0.0359 │ 0.0377    
  MACVO@v2_H000 │ 0.2499 │ 0.125 │ 0.2794   │ 0.0106 │ 0.0351 │ 0.0367   │ 0.1497 │ 0.3183 │ 0.3518   │ 0.0114 │ 0.0359 │ 0.0377    
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number:  v1.31.1 / v1.28.0 
* Python version: 3.11.11 
* operating system: Ubuntu 24 LTS
* did you change the source code? : no 
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "map_tile_api_token": "",
    "map_tile_provider": "OpenStreetMap.Mapnik",
    "plot_3d_zoom": 0.9,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        10,
        10
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_legend_loc": "best",
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_enabled": true,
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

Also attaching the zip file containing poses.npy (est trajectory) and ref_poses.npy (reference trajectory), where both are Nx8 np.array with format [time, x, y, z, q_x, q_y, q_z, q_w]

MichaelGrupp, 2025-04-07:
Probably this here is causing it:

```python
                          align_origin=True,
                          align=True,
```

Using both at the same time isn't possible (also not in the older version, it only uses one of them, you can see this also in the title string of the result). If you set `align_origin=False` it should be like before for you.

There was a change in how the `evo_ape/rpe` command line tool allows mixing of alignment flags: https://github.com/MichaelGrupp/evo/commit/8f0d639b5a405bd710c9d812dab16c14299ff28d (see also related discussion https://github.com/MichaelGrupp/evo/issues/693#issuecomment-2411580917)

The mutual exclusion of the `--align` and `--align_origin` flags is handled on the command line parser level of those apps, but that's not affecting you since you're importing the function from there directly in your code.

MarkChenYutian, 2025-04-07:
Thanks for response! Just want to make sure that I should use `align = True` and `align_origin = False` to produce the results aligned with 1.28.0's output.

closed 2025-04-12 by MichaelGrupp